yubox-node-org/AsyncTCPSock #8: Redefinitions when using with ESPAsyncWebServer

davidkleng, 2021-11-19:
Hi, when I try to use both libs im getting errors about redefinition of functions. Here is a copy of the console.

```
.pio\build\esp32dev\lib9da\libAsyncTCP.a(AsyncTCP.cpp.o): In function `AsyncClient::connect(IPAddress, unsigned short)':
c:\users\david\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0/functional:2077: multiple definition of `AsyncClient::connect(IPAddress, unsigned short)'
.pio\build\esp32dev\libe5f\libAsyncTCPSock.a(AsyncTCP.cpp.o):c:\users\david\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\bits/move.h:188: first defined here
.pio\build\esp32dev\lib9da\libAsyncTCP.a(AsyncTCP.cpp.o): In function `AsyncClient::connect(char const*, unsigned short)':
AsyncTCP.cpp:(.text._ZN11AsyncClient7connectEPKct+0x0): multiple definition of `AsyncClient::connect(char const*, unsigned short)'
.pio\build\esp32dev\libe5f\libAsyncTCPSock.a(AsyncTCP.cpp.o):AsyncTCP.cpp:(.text._ZN11AsyncClient7connectEPKct+0x0): first defined here
.pio\build\esp32dev\lib9da\libAsyncTCP.a(AsyncTCP.cpp.o): In function `AsyncClient::abort()':
AsyncTCP.cpp:(.text._ZN11AsyncClient5abortEv+0x0): multiple definition of `AsyncClient::abort()'
.pio\build\esp32dev\libe5f\libAsyncTCPSock.a(AsyncTCP.cpp.o):AsyncTCP.cpp:(.text._ZN11AsyncClient5abortEv+0x0): first defined here
.pio\build\esp32dev\lib9da\libAsyncTCP.a(AsyncTCP.cpp.o): In function `AsyncClient::space()':
AsyncTCP.cpp:(.text._ZN11AsyncClient5spaceEv+0x0): multiple definition of `AsyncClient::space()'
.pio\build\esp32dev\libe5f\libAsyncTCPSock.a(AsyncTCP.cpp.o):AsyncTCP.cpp:(.text._ZN11AsyncClient5spaceEv+0x0): first defined here
.pio\build\esp32dev\lib9da\libAsyncTCP.a(AsyncTCP.cpp.o): In function `AsyncClient::add(char const*, unsigned int, unsigned char)':
AsyncTCP.cpp:(.text._ZN11AsyncClient3addEPKcjh+0x0): multiple definition of `AsyncClient::add(char const*, unsigned int, unsigned char)'
.pio\build\esp32dev\libe5f\libAsyncTCPSock.a(AsyncTCP.cpp.o):AsyncTCP.cpp:(.text._ZN11AsyncClient3addEPKcjh+0x0): first defined here
.pio\build\esp32dev\lib9da\libAsyncTCP.a(AsyncTCP.cpp.o): In function `AsyncClient::send()':
AsyncTCP.cpp:(.text._ZN11AsyncClient4sendEv+0x0): multiple definition of `AsyncClient::send()'
.pio\build\esp32dev\libe5f\libAsyncTCPSock.a(AsyncTCP.cpp.o):AsyncTCP.cpp:(.text._ZN11AsyncClient4sendEv+0x0): first defined here
.pio\build\esp32dev\lib9da\libAsyncTCP.a(AsyncTCP.cpp.o): In function `AsyncClient::free()':
AsyncTCP.cpp:(.text._ZN11AsyncClient4freeEv+0x0): multiple definition of `AsyncClient::free()'
.pio\build\esp32dev\libe5f\libAsyncTCPSock.a(AsyncTCP.cpp.o):AsyncTCP.cpp:(.text._ZN11AsyncClient4freeEv+0x0): first defined here
.pio\build\esp32dev\lib9da\libAsyncTCP.a(AsyncTCP.cpp.o): In function `AsyncClient::write(char const*, unsigned int, unsigned char)':
AsyncTCP.cpp:(.text._ZN11AsyncClient5writeEPKcjh+0x0): multiple definition of `AsyncClient::write(char const*, unsigned int, unsigned char)'
.pio\build\esp32dev\libe5f\libAsyncTCPSock.a(AsyncTCP.cpp.o):AsyncTCP.cpp:(.text._ZN11AsyncClient5writeEPKcjh+0x0): first defined here
.pio\build\esp32dev\lib9da\libAsyncTCP.a(AsyncTCP.cpp.o): In function `AsyncClient::write(char const*)':
AsyncTCP.cpp:(.text._ZN11AsyncClient5writeEPKc+0x0): multiple definition of `AsyncClient::write(char const*)'
.pio\build\esp32dev\libe5f\libAsyncTCPSock.a(AsyncTCP.cpp.o):AsyncTCP.cpp:(.text._ZN11AsyncClient5writeEPKc+0x0): first defined here
.pio\build\esp32dev\lib9da\libAsyncTCP.a(AsyncTCP.cpp.o): In function `AsyncClient::setRxTimeout(unsigned int)':
AsyncTCP.cpp:(.text._ZN11AsyncClient12setRxTimeoutEj+0x0): multiple definition of `AsyncClient::setRxTimeout(unsigned int)'
.pio\build\esp32dev\libe5f\libAsyncTCPSock.a(AsyncTCP.cpp.o):AsyncTCP.cpp:(.text._ZN11AsyncClient12setRxTimeoutEj+0x0): first defined here
.pio\build\esp32dev\lib9da\libAsyncTCP.a(AsyncTCP.cpp.o): In function `AsyncClient::getRxTimeout()':
AsyncTCP.cpp:(.text._ZN11AsyncClient12getRxTimeoutEv+0x0): multiple definition of `AsyncClient::getRxTimeout()'
.pio\build\esp32dev\libe5f\libAsyncTCPSock.a(AsyncTCP.cpp.o):AsyncTCP.cpp:(.text._ZN11AsyncClient12getRxTimeoutEv+0x0): first defined here
.pio\build\esp32dev\lib9da\libAsyncTCP.a(AsyncTCP.cpp.o): In function `AsyncClient::getAckTimeout()':
AsyncTCP.cpp:(.text._ZN11AsyncClient13getAckTimeoutEv+0x0): multiple definition of `AsyncClient::getAckTimeout()'
.pio\build\esp32dev\libe5f\libAsyncTCPSock.a(AsyncTCP.cpp.o):AsyncTCP.cpp:(.text._ZN11AsyncClient13getAckTimeoutEv+0x0): first defined here
.pio\build\esp32dev\lib9da\libAsyncTCP.a(AsyncTCP.cpp.o): In function `AsyncClient::setAckTimeout(unsigned int)':
AsyncTCP.cpp:(.text._ZN11AsyncClient13setAckTimeoutEj+0x0): multiple definition of `AsyncClient::setAckTimeout(unsigned int)'
.pio\build\esp32dev\libe5f\libAsyncTCPSock.a(AsyncTCP.cpp.o):AsyncTCP.cpp:(.text._ZN11AsyncClient13setAckTimeoutEj+0x0): first defined here
.pio\build\esp32dev\lib9da\libAsyncTCP.a(AsyncTCP.cpp.o): In function `AsyncClient::setNoDelay(bool)':
AsyncTCP.cpp:(.text._ZN11AsyncClient10setNoDelayEb+0x0): multiple definition of `AsyncClient::setNoDelay(bool)'
.pio\build\esp32dev\libe5f\libAsyncTCPSock.a(AsyncTCP.cpp.o):AsyncTCP.cpp:(.text._ZN11AsyncClient10setNoDelayEb+0x0): first defined here
.pio\build\esp32dev\lib9da\libAsyncTCP.a(AsyncTCP.cpp.o): In function `AsyncClient::getNoDelay()':
AsyncTCP.cpp:(.text._ZN11AsyncClient10getNoDelayEv+0x0): multiple definition of `AsyncClient::getNoDelay()'
.pio\build\esp32dev\libe5f\libAsyncTCPSock.a(AsyncTCP.cpp.o):AsyncTCP.cpp:(.text._ZN11AsyncClient10getNoDelayEv+0x0): first defined here
.pio\build\esp32dev\lib9da\libAsyncTCP.a(AsyncTCP.cpp.o): In function `AsyncClient::getRemoteAddress()':
AsyncTCP.cpp:(.text._ZN11AsyncClient16getRemoteAddressEv+0x0): multiple definition of `AsyncClient::getRemoteAddress()'
.pio\build\esp32dev\libe5f\libAsyncTCPSock.a(AsyncTCP.cpp.o):AsyncTCP.cpp:(.text._ZN11AsyncClient16getRemoteAddressEv+0x0): first defined here
.pio\build\esp32dev\lib9da\libAsyncTCP.a(AsyncTCP.cpp.o): In function `AsyncClient::getRemotePort()':
AsyncTCP.cpp:(.text._ZN11AsyncClient13getRemotePortEv+0x0): multiple definition of `AsyncClient::getRemotePort()'
.pio\build\esp32dev\libe5f\libAsyncTCPSock.a(AsyncTCP.cpp.o):AsyncTCP.cpp:(.text._ZN11AsyncClient13getRemotePortEv+0x0): first defined here
.pio\build\esp32dev\lib9da\libAsyncTCP.a(AsyncTCP.cpp.o): In function `AsyncClient::getLocalAddress()':
AsyncTCP.cpp:(.text._ZN11AsyncClient15getLocalAddressEv+0x0): multiple definition of `AsyncClient::getLocalAddress()'
.pio\build\esp32dev\libe5f\libAsyncTCPSock.a(AsyncTCP.cpp.o):AsyncTCP.cpp:(.text._ZN11AsyncClient15getLocalAddressEv+0x0): first defined here
.pio\build\esp32dev\lib9da\libAsyncTCP.a(AsyncTCP.cpp.o): In function `AsyncClient::getLocalPort()':
AsyncTCP.cpp:(.text._ZN11AsyncClient12getLocalPortEv+0x0): multiple definition of `AsyncClient::getLocalPort()'
.pio\build\esp32dev\libe5f\libAsyncTCPSock.a(AsyncTCP.cpp.o):AsyncTCP.cpp:(.text._ZN11AsyncClient12getLocalPortEv+0x0): first defined here
.pio\build\esp32dev\lib9da\libAsyncTCP.a(AsyncTCP.cpp.o): In function `AsyncClient::remoteIP()':
AsyncTCP.cpp:(.text._ZN11AsyncClient8remoteIPEv+0x0): multiple definition of `AsyncClient::remoteIP()'
.pio\build\esp32dev\libe5f\libAsyncTCPSock.a(AsyncTCP.cpp.o):AsyncTCP.cpp:(.text._ZN11AsyncClient8remoteIPEv+0x0): first defined here
.pio\build\esp32dev\lib9da\libAsyncTCP.a(AsyncTCP.cpp.o): In function `AsyncClient::remotePort()':
AsyncTCP.cpp:(.text._ZN11AsyncClient10remotePortEv+0x0): multiple definition of `AsyncClient::remotePort()'
.pio\build\esp32dev\libe5f\libAsyncTCPSock.a(AsyncTCP.cpp.o):AsyncTCP.cpp:(.text._ZN11AsyncClient10remotePortEv+0x0): first defined here
.pio\build\esp32dev\lib9da\libAsyncTCP.a(AsyncTCP.cpp.o): In function `AsyncClient::localIP()':
AsyncTCP.cpp:(.text._ZN11AsyncClient7localIPEv+0x0): multiple definition of `AsyncClient::localIP()'
.pio\build\esp32dev\libe5f\libAsyncTCPSock.a(AsyncTCP.cpp.o):AsyncTCP.cpp:(.text._ZN11AsyncClient7localIPEv+0x0): first defined here
.pio\build\esp32dev\lib9da\libAsyncTCP.a(AsyncTCP.cpp.o): In function `AsyncClient::localPort()':
[... 52 lines cut ...]
.pio\build\esp32dev\libe5f\libAsyncTCPSock.a(AsyncTCP.cpp.o):AsyncTCP.cpp:(.text._ZN11AsyncClient9onConnectESt8functionIFvPvPS_EES1_+0x0): first defined here
.pio\build\esp32dev\lib9da\libAsyncTCP.a(AsyncTCP.cpp.o): In function `AsyncClient::onDisconnect(std::function<void (void*, AsyncClient*)>, void*)':
[... cut ...]
.pio\build\esp32dev\lib9da\libAsyncTCP.a(AsyncTCP.cpp.o): In function `AsyncClient::onTimeout(std::function<void (void*, AsyncClient*, unsigned int)>, void*)':
AsyncTCP.cpp:(.text._ZN11AsyncClient9onTimeoutESt8functionIFvPvPS_jEES1_+0x0): multiple definition of `AsyncClient::onTimeout(std::function<void (void*, AsyncClient*, unsigned int)>, void*)'
.pio\build\esp32dev\libe5f\libAsyncTCPSock.a(AsyncTCP.cpp.o):AsyncTCP.cpp:(.text._ZN11AsyncClient9onTimeoutESt8functionIFvPvPS_jEES1_+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1
```

avillacis, 2021-11-21:
> Hi, when I try to use both libs im getting errors about redefinition of functions.
This situation is completely expected and deliberate.

This library (AsyncTCPSock) is supposed to be a *reimplementation* of the AsyncTCP API. As such, it is supposed to define all public methods that were defined by the original AsyncTCP library. The two libraries are *not* supposed to be used together in the same project. You should choose one of them and delete/move aside the other.

Is the presence of both libraries in the same project accidental? If not, why is it required to attempt use of both libraries at the same time in the same project?

davidkleng, 2021-11-21:
Hi, im currently using Plattformio and no, i didn't include both libs, only AsyncTCPSock. Don't know why, but when I compile the code Plattformio downloads both libs and this error pops. Here are the libs im currently using: 

```
board_upload.flash_size = 16MB
board_upload.maximum_size = 16777216
extra_scripts = replace_fs.py
board_build.filesystem = littlefs
lib_deps = 
    ;https://github.com/me-no-dev/AsyncTCP
    ;https://github.com/me-no-dev/ESPAsyncWebServer    
    https://github.com/lorol/LITTLEFS
    
    ;https://github.com/4-20ma/ModbusMaster.git
    https://github.com/adafruit/Adafruit_ADS1X15.git
    https://github.com/adafruit/RTClib.git	
	https://github.com/taranais/NTPClient.git
    https://github.com/PaulStoffregen/OneWire.git
    https://github.com/xreef/PCF8574_library
    https://github.com/yaacov/ArduinoModbusSlave

    https://github.com/yubox-node-org/AsyncTCPSock
	https://github.com/yubox-node-org/ESPAsyncWebServer

```

I have already tryed deleting the .pio folder and cleaning the project. No mathers what when I try to compile it, plattformio dowloads the lib again.

proddy, 2021-11-21:
It's probably because ESPAsyncWebServer has a dependency on that library. When building you'll see the library tree. Try excluding it using `lib-ignore` https://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-ignore

davidkleng, 2021-11-21:
Thanks, i will try it. I got it to complile manually deleting the redefinitions. But when it did work, it gave me otrher errors like this Issue i posted:

[https://github.com/yubox-node-org/ESPAsyncWebServer/issues/3](url)

When you look at the backtrace you can see that is using methods from AsyncTCPSock and AsyncTCP, may that be the problem?

Thanks for your help.

proddy, 2021-11-21:
it's just a wrapper, so backwards compatible. Is your project on GitHub?

davidkleng, 2021-11-21:
Sorry, it is not. But i may be able to share some code, do you need to seesomething?

davidkleng, 2021-12-01:
Adding 
```
lib_ignore =
    AsyncTCP
```

to platformio.ini solved the problem. Thanks.

closed 2021-12-01 by davidkleng