yubox-node-org/AsyncTCPSock #11: mbedtls_ssl_handshake() blocks in 2nd call because of MBEDTLS_ERR_SSL_WANT_READ

BlueAndi, 2022-01-07:
Currently 100% reproducable in my local setup:
```
        41 INFO             InitState.cpp:  353 ESP32 chip rev.: 1
        46 INFO             InitState.cpp:  354 ESP32 SDK version: v3.3.5-1-g85c43024c
        61 INFO             InitState.cpp:  356 LwIP version: 2.0.3
```

Log output:

```
[V][AsyncTCP.cpp:563] connect(): connect to api.github.com port 443 using DNS...
[V][AsyncTCP.cpp:574] connect():        waiting for DNS resolution
[V][AsyncTCP_TLS_Context.cpp:117] _startSSLClient(): Seeding the random number generator
[V][AsyncTCP_TLS_Context.cpp:126] _startSSLClient(): Setting up the SSL/TLS structure...
[I][AsyncTCP_TLS_Context.cpp:139] _startSSLClient(): WARNING: Skipping SSL Verification. INSECURE!
[V][AsyncTCP_TLS_Context.cpp:212] _startSSLClient(): Setting hostname for TLS session...
[I][AsyncTCP.cpp:709] _sockIsWriteable(): _runSSLHandshakeLoop() begin
[I][AsyncTCP.cpp:711] _sockIsWriteable(): _runSSLHandshakeLoop() end
[I][AsyncTCP.cpp:715] _sockIsWriteable(): ASYNCTCP_TLS_CAN_RETRY
[I][AsyncTCP.cpp:709] _sockIsWriteable(): _runSSLHandshakeLoop() begin

Never returns in 2nd call of _runSSLHandshakeLoop(), because only the first call of mbedtls_ssl_handshake() seems to be non-blocking. The first call of mbedtls_ssl_handshake() returns MBEDTLS_ERR_SSL_WANT_READ.

This causes a task watchdog reset as result.
```

https://github.com/yubox-node-org/AsyncTCPSock/blob/f22b097b2ed7f213149b9c69d0d8b1b7a1b5732a/src/AsyncTCP_TLS_Context.cpp#L226

is used, but according to

https://github.com/espressif/arduino-esp32/blob/caef4006af491130136b219c1205bdcf8f08bf2b/tools/sdk/esp32/include/mbedtls/mbedtls/include/mbedtls/ssl.h#L1391-L1425
```C
/**
 * \brief          Set the underlying BIO callbacks for write, read and
 *                 read-with-timeout.
 *
 * \param ssl      SSL context
 * \param p_bio    parameter (context) shared by BIO callbacks
 * \param f_send   write callback
 * \param f_recv   read callback
 * \param f_recv_timeout blocking read callback with timeout.
 *
 * \note           One of f_recv or f_recv_timeout can be NULL, in which case
 *                 the other is used. If both are non-NULL, f_recv_timeout is
 *                 used and f_recv is ignored (as if it were NULL).
 *
 * \note           The two most common use cases are:
 *                 - non-blocking I/O, f_recv != NULL, f_recv_timeout == NULL
 *                 - blocking I/O, f_recv == NULL, f_recv_timout != NULL
 *
 * \note           For DTLS, you need to provide either a non-NULL
 *                 f_recv_timeout callback, or a f_recv that doesn't block.
 *
 * \note           See the documentations of \c mbedtls_ssl_send_t,
 *                 \c mbedtls_ssl_recv_t and \c mbedtls_ssl_recv_timeout_t for
 *                 the conventions those callbacks must follow.
 *
 * \note           On some platforms, net_sockets.c provides
 *                 \c mbedtls_net_send(), \c mbedtls_net_recv() and
 *                 \c mbedtls_net_recv_timeout() that are suitable to be used
 *                 here.
 */
void mbedtls_ssl_set_bio( mbedtls_ssl_context *ssl,
                          void *p_bio,
                          mbedtls_ssl_send_t *f_send,
                          mbedtls_ssl_recv_t *f_recv,
                          mbedtls_ssl_recv_timeout_t *f_recv_timeout );
 ```

for DTLS a non-blocking ```mbedtls_ssl_recv_t *f_recv``` is necessary or using ```mbedtls_ssl_recv_timeout_t *f_recv_timeout```, like the ```mbedtls_net_recv_timeout()```.

Using ```mbedtls_net_recv()``` will block, but the socket is set to NON-BLOCKING, see here:
https://github.com/yubox-node-org/AsyncTCPSock/blob/f22b097b2ed7f213149b9c69d0d8b1b7a1b5732a/src/AsyncTCP.cpp#L510
Which means it should work, but didn't. It depends on the ```fcntl()``` implementation, which is lwip specific.

I checked now the function in lwip (branch 2.0.3-esp):
https://github.com/espressif/esp-lwip/blob/2acf959a2bb559313cd2bf9306c24612ba3d0e19/src/api/sockets.c#L3177-L3182
```C
/** A minimal implementation of fcntl.
 * Currently only the commands F_GETFL and F_SETFL are implemented.
 * Only the flag O_NONBLOCK is implemented.
 */
int
lwip_fcntl(int s, int cmd, int val)
```
and it states, that only O_NONBLOCK is supported for F_SETFL. Hmm ... strange.

What I did now is to call ```mbedtls_ssl_set_bio()``` additional with ```mbedtls_net_recv_timeout```.
```C
mbedtls_ssl_set_bio(&ssl_ctx, &_socket, mbedtls_net_send, mbedtls_net_recv, mbedtls_net_recv_timeout);
```
and set the timeout with ```mbedtls_ssl_conf_read_timeout()``` right after this statement
https://github.com/yubox-node-org/AsyncTCPSock/blob/f22b097b2ed7f213149b9c69d0d8b1b7a1b5732a/src/AsyncTCP_TLS_Context.cpp#L128-L133

Here the log output again and without task watchdog:

```
[V][AsyncTCP.cpp:563] connect(): connect to api.github.com port 443 using DNS...
[V][AsyncTCP.cpp:574] connect():        waiting for DNS resolution
[V][AsyncTCP_TLS_Context.cpp:117] _startSSLClient(): Seeding the random number generator
[V][AsyncTCP_TLS_Context.cpp:126] _startSSLClient(): Setting up the SSL/TLS structure...
[I][AsyncTCP_TLS_Context.cpp:139] _startSSLClient(): WARNING: Skipping SSL Verification. INSECURE!
[V][AsyncTCP_TLS_Context.cpp:212] _startSSLClient(): Setting hostname for TLS session...
[I][AsyncTCP.cpp:709] _sockIsWriteable(): _runSSLHandshakeLoop() begin
[E][AsyncTCP_TLS_Context.cpp:28] _handle_error(): [runSSLHandshake():247]: (-26624) SSL - The operation timed out
[E][AsyncTCP.cpp:641] _runSSLHandshakeLoop(): TLS setup failed with error -26624, closing socket...
[V][AsyncTCP_TLS_Context.cpp:336] ~AsyncTCP_TLS_Context(): Cleaning SSL connection.
[I][AsyncTCP.cpp:711] _sockIsWriteable(): _runSSLHandshakeLoop() end
[I][AsyncTCP.cpp:715] _sockIsWriteable(): ASYNCTCP_TLS_CAN_RETRY
```

It is working now and there is no task watchdog anymore.
I fear the problem may be down in lwip/arduino32 ... what do you mean?

avillacis, 2022-01-10:
My test program for the SSL functionality pointed to www.eff.org and used a Lets Encrypt certificate. Is the issue also reproducible with that server?

My test setup is Arduino-ESP32 2.0.2 which uses Espressif 4.4 . Have you tried updating your setup?

BlueAndi, 2022-01-12:
I will setup my project on the latest Arduino framework 2..0.2 and try to reproduce it again. I will come back with feedback.

BlueAndi, 2022-02-27:
Just a short life sign ... didn't manage to test it yet, need a little bit more time.

BlueAndi, 2022-04-02:
Currently I don't get Arduino framework 2.x running with pio. I will wait until pio releases the support official.
I will close this issue and come back if I can reproduce it in 2.x

closed 2022-04-02 by BlueAndi