yubox-node-org/AsyncTCPSock #16: InstrFetchProhibited crash on WiFi functions when using this library.

zekageri, 2022-11-27:
Board crashes on wifi instruction calls.

I have tried with WiFi.softAPConfig(apIP, apIP, netMsk);
And WiFi.mode(WIFI_STA);
Even on ethernet cb setup.

**Crash:**

```
[  2246][D][WiFiGeneric.cpp:929] _eventCallback(): Arduino Event: 0 - WIFI_READY
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x0c004136  PS      : 0x00060930  A0      : 0x800e1d78  A1      : 0x3ffb2360  
A2      : 0x00000000  A3      : 0x401a41e8  A4      : 0x3ffb2470  A5      : 0x00000000  
A6      : 0x3ffb8a28  A7      : 0x00000000  A8      : 0x800e1d3e  A9      : 0x3ffb2300  
A10     : 0x3ffb2470  A11     : 0x401a41e8  A12     : 0x00000002  A13     : 0x00000001  
A14     : 0x00060b20  A15     : 0x00000001  SAR     : 0x00000020  EXCCAUSE: 0x00000014
EXCVADDR: 0x0c004134  LBEG    : 0x4008bd2d  LEND    : 0x4008bd3d  LCOUNT  : 0xfffffffd

Backtrace:0x0c004133:0x3ffb23600x400e1d75:0x3ffb2380 0x400e1e2d:0x3ffb23a0 0x400e232c:0x3ffb2530 

  #0  0x0c004133:0x3ffb2360 in ?? ??:0
  #1  0x400e1d75:0x3ffb2380 in WiFiEventCbList::WiFiEventCbList(WiFiEventCbList const&) at C:/Users/zekag/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:740
  #2  0x400e1e2d:0x3ffb23a0 in WiFiGenericClass::_eventCallback(arduino_event_t*) at C:/Users/zekag/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:1050
  #3  0x400e232c:0x3ffb2530 in _arduino_event_task(void*) at C:/Users/zekag/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:303

ELF file SHA256: 0000000000000000

Rebooting...
```

Crash does not happen with [https://github.com/khoih-prog/AsyncTCP_SSL](https://github.com/khoih-prog/AsyncTCP_SSL)
not with the official Async tcp library.

Sketch is the same in every case.

avillacis, 2022-11-28:
Do you have a minimal sketch that will reproduce this crash?

Could you please specify the branches and commit tips at which you downloaded snapshots of the libraries?

Does your crashing sketch require SSL support?

Does the crash occur on sketch startup? Or does it occur when handling an incoming http request?

At which point in the lifetime of the sketch does it attempt to call `begin()` on the `AsyncWebServer` instance? Right in the body of the `setup()` function? Inside one of the wifi callbacks? Or some other way?

zekageri, 2022-11-29:
Sorry for not including a sketch.
I will make one today.

The sketch does not reach server begin, it crashes as soon as I touch any built in WiFi method and crashes with the same error no matter what method I call.

It does not require SSL and I do not initialize any SSL related stuff.

Both of the libraries are from the main brach.