yubox-node-org/AsyncTCPSock #18: tcp_process error every 1-2 or 3 days.

zekageri, 2023-01-05:
I have a strange error. My application runs flawlessly but after 1 or 2 or max 3 days it crashes.
My device is in a remote location, it was really hard to get the crash dump.
I have no ram loss and the application does not notify with anything before the crash.

**Here is the crash dump:**

```cpp
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x401a68d3  PS      : 0x00060230  A0      : 0x801a696b  A1      : 0x3ffb42f0  
A2      : 0x00000000  A3      : 0x3ffdfbec  A4      : 0x00000000  A5      : 0x3ffcbfc0  
A6      : 0x3ffcbfd8  A7      : 0x00060023  A8      : 0x8014499c  A9      : 0x3ffb42e0  
A10     : 0x00000000  A11     : 0x00000006  A12     : 0x00000014  A13     : 0x0000ffff  
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000001c  LBEG    : 0x4008b014  LEND    : 0x4008b02a  LCOUNT  : 0xffffffff  

Backtrace: 0x401a68d0:0x3ffb42f0 0x401a6968:0x3ffb4320 0x40138675:0x3ffb4340 0x4013d6f2:0x3ffb43b0 0x40141766:0x3ffb43e0 0x4013220d:0x3ffb4400

  #0  0x401a68d0:0x3ffb42f0 in AsyncServer::_accept(tcp_pcb*, signed char) at lib/AsyncTCP/src/AsyncTCP.cpp:1322
  #1  0x401a6968:0x3ffb4320 in AsyncServer::_s_accept(void*, tcp_pcb*, signed char) at lib/AsyncTCP/src/AsyncTCP.cpp:1361
  #2  0x40138675:0x3ffb4340 in tcp_process at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/tcp_in.c:945
      (inlined by) tcp_input at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/tcp_in.c:438
  #3  0x4013d6f2:0x3ffb43b0 in ip4_input at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/core/ipv4/ip4.c:800
  #4  0x40141766:0x3ffb43e0 in ethernet_input at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/netif/ethernet.c:186
  #5  0x4013220d:0x3ffb4400 in tcpip_thread_handle_msg at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/tcpip.c:180
      (inlined by) tcpip_thread at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/lwip/lwip/src/api/tcpip.c:154

ELF file SHA256: 225259e45e21a87e

Rebooting...
ets Jul 29 2019 12:21:46
```

And here are the relevant functions:

```cpp
// Sketch uses ETH.h mostly.
// Crash happens in the following funcions:

int8_t AsyncServer::_s_accept(void * arg, tcp_pcb * pcb, int8_t err){
    return reinterpret_cast<AsyncServer*>(arg)->_accept(pcb, err);
}

//runs on LwIP thread
int8_t AsyncServer::_accept(tcp_pcb* pcb, int8_t err){
    //ets_printf("+A: 0x%08x\n", pcb);
    if(_connect_cb){
        AsyncClient *c = new AsyncClient(pcb);
        if(c){
            c->setNoDelay(_noDelay);
            return _tcp_accept(this, c);
        }
    }
    if(tcp_close(pcb) != ERR_OK){
        tcp_abort(pcb);
    }
    log_e("FAIL");
    return ERR_OK;
}

//Used to switch out from LwIP thread
static int8_t _tcp_accept(void * arg, AsyncClient * client) {
    lwip_event_packet_t * e = (lwip_event_packet_t *)malloc(sizeof(lwip_event_packet_t));
    if(!e) return ERR_MEM;
    e->event = LWIP_TCP_ACCEPT;
    e->arg = arg;
    e->accept.client = client;
    if (!_prepend_async_event(&e)) {
        free((void*)(e));
    }
    return ERR_OK;
}
```

Pablo2048, 2023-01-05:
Wrong repo - it seems like you are using AsyncTCP library, not this one...

zekageri, 2023-01-05:
Oh god. You are right. I must have left the old library in the libs folder instead of `AsyncTCPSock`...
Sorry. Will test it further with `AsyncTCPSock`. :'(

closed 2023-01-05 by zekageri