yubox-node-org/AsyncTCPSock #23: AsyncTCPSock : dead ? releases ?

mathieucarbou, 2024-10-13:
Hello,

With some community help I am maintaining [ESPAsyncWebServer](https://github.com/mathieucarbou/ESPAsyncWebServer).

As you can see in the README, I am explaining how to switch AsyncTCP to AsyncTCPSock.

But there are several problems:

1) AsyncTCPSock is not deployed in Arduino Registry neither PlatformIO regstry
2) AsyncTCPSock has no releases
3) Some Prs are waiting to be merged for Arduino 3 support.
4) Macros configurations are not well aligned with [AsyncTCP](https://github.com/mathieucarbou/AsyncTCP)

Is this project dead ?

For the moment, I decided to fork it, and merge the PR for arduino 3, plus do some macro config alignment, retaining backward compatibility, and issue some releases.

But it would be nice to have it deployed also. 

I can do all that too (deploying to Arduino Registry and PlatformIO), retaining you as the author and adding me as a maintainer.

But I first wanted to ask what is the project state.

Thanks!

DRSDavidSoft, 2024-10-20:
I was wondering the same thing too. Very interested to know about the results! 👍🏻

mathieucarbou, 2024-10-20:
Fo those who need, I took the liberty to fork the project, applies some PR fixes and release (tag) my dev branch.

https://github.com/mathieucarbou/AsyncTCPSock/releases

These releases can be used in the community maintained fork of [ESPAsyncWebServer](https://github.com/mathieucarbou/ESPAsyncWebServer) as explained in the README.